gbmhunter/AltiumScriptCentral #173: Main.vbs "Wrong number of arguments or invalid property assignment:'StdErr'

Barleyman, 2015-01-15:
As it says. This happens running main.vbs. 14.1. v22.3.2.0. Both Altium v14 and v15 suffer of both problems. 

[image]

After that the menu comes up and I reselect the footprint I was working on and select "renumber pads" and boom, exceoption in scriptingsystem.dll.dll

[image]

Perhaps I'm just being stupid and don't understand the correct procedure. Open project, run script and pick "main"..

gbmhunter, 2015-01-15:
Thanks for noting! Your doing things correctly, this I my fault, and it should be fixed in the latest version (v22.3.4.0). Let me know if this works for you and then this issue can be closed.

gbmhunter, 2015-01-22:
Fixed.

closed 2015-01-22 by gbmhunter