gbmhunter/AltiumScriptCentral #204: Draw Polygon. Error parse dot in number.

sledgeh, 2015-09-20:
[image]
This is because my localized separator is comma. All works normally with it. Since Altium supports both separators it should be cool to make script "eat" both too.

sledgeh, 2015-09-20:
I have solved it with 

    ...
    ' Line thickness
    ' check locale
    if( CStr(0.1) = "0,1" ) then
        EditLineThicknessMm.Text = (Replace(EditLineThicknessMm.Text,".",","))
    end if
    If Not IsPerfectlyNumeric(EditLineThicknessMm.Text) Then
    ...

gbmhunter, 2015-09-24:
Thanks for reporting this bug and presenting a solution! I will add it to the code soon.

gbmhunter, 2015-09-25:
Weird, I tried finding the exact reason why `IsPerfectlyNumeric()` function in Utils.vbs would be returning false, as all the tests inside that function are meant to be locale-aware (i.e. understand that you use a comma for the decimal separator).

I tried change my locale to something that uses a comma with `SetLocale(3079)` (this changes the locale to Austria), however I couldn't get the code to recognise the comma.

@sledgeh, would you be able to determine what part of the `IsPerfectlyNumeric()` function in Utils.vbs is returning false? There are three if statements in that function.

sledgeh, 2015-09-25:
I will.

sledgeh, 2015-09-25:
`IsNumeric("1.2")` returns `false`

gbmhunter, 2015-09-27:
`IsNumeric("1.2") returns `true` for me, presumably because my locale supports the period the `.` while yours supports the comma (`,`).

However, this still supports the theory that the `IsPerfectlyNumeric()` function should work correctly for both locales. Would you be able to provide info on which part of `IsPerfectlyNumeric()` returns `false` for a number such as `1,2` (note that's a comma, not a period as you mentioned above) on your system.

sledgeh, 2015-09-28:
> which part of `IsPerfectlyNumeric()`...  

I supposed my answer was clear.
> `IsNumeric("1.2")` returns false

And since it is part of `IsPerfectlyNumeric()`, `IsPerfectlyNumeric()` returns false.

    showmessage IsPerfectlyNumeric("1")          'true
    showmessage IsPerfectlyNumeric("1.2")        'false
    showmessage IsPerfectlyNumeric("1,2")        'true
    showmessage IsNumeric("1.2")                 'false

sledgeh, 2015-10-09:
What about my pull request? It provides valid fixes.